GyverLibs/GyverIO #3: Не компилируется на ESP32-C6 с ядром 3.0.1

amirotin, 2024-06-17:
Компилирую на platformio с ядром 3.0.1 и IDF 5.1
```
.pio/libdeps/xiao-esp32-c6/GyverIO/src/gio/gio_esp32.h: In function 'void gio::mode(uint8_t, uint8_t)':
.pio/libdeps/xiao-esp32-c6/GyverIO/src/gio/gio_esp32.h:35:57: error: no match for 'operator=' (operand types are 'volatile gpio_enable_w1tc_reg_t' and 'long unsigned int')
   35 |             if (pin < 32) GPIO.enable_w1tc = (1ul << pin);
      |                                                         ^
```

```
.pio/libdeps/xiao-esp32-c6/GyverIO/src/gio/gio_esp32.h: In function 'int gio::read(uint8_t)':
.pio/libdeps/xiao-esp32-c6/GyverIO/src/gio/gio_esp32.h:64:35: error: no match for 'operator>>' (operand types are 'volatile gpio_in_reg_t' and 'uint8_t' {aka 'unsigned char'})
   64 |     if (pin < 32) return (GPIO.in >> pin) & 0x1;
      |                           ~~~~~~~ ^~ ~~~
      |                                |     |
      |                                |     uint8_t {aka unsigned char}
      |                                volatile gpio_in_reg_t
```

GyverLibs, 2024-06-17:
не вижу поддержку arduino фреймворка в platformio для C6

GyverLibs, 2024-06-17:
я и не говорю что проблема в платформио. Как мне повторить баг, как установить ядро? Этих двух ссылок мало

amirotin, 2024-06-17:
> я и не говорю что проблема в платформио. Как мне повторить баг, как установить ядро? Этих двух ссылок мало

Самый простой вариант, чтобы не устраивать танцы с бубном в платформио, это протестить в Arduino IDE:
1. Создаем пустой скетч
2. Выбираем в качестве платы любую ESP32 C6 (в моем случае Seeed Studio Xiao ESP32C6, но ошибка на любой плате С6)
3. Добавляем библиотеку GyverIO
4. Нажимаем компиляцию

GyverLibs, 2024-06-17:
Я понимаю что проще через иде2. Но она никак не поможет исправить багу и вообще покопаться до корня ошибки

amirotin, 2024-06-17:
Тогда для платформио шаги такие:

1. В папку .platformio\boards добавить файл xiao-esp32-c6.json
2. Создаем проект на любой плате с поддержкой Arduino (например S3). Это из-за того, что в платформио нет шаблона проекта для C6 на ардуино. Или можно уже имеющийся на ардуино изменить
3. platformio.ini указываем вот такие настройки
```
[env:xiao-esp32-c6]
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino
board = xiao-esp32-c6
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.1
	framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.1/esp32-arduino-libs-3.0.1.zip
```
4. Дальше уже как обычно добавляем библиотеку и компилируем

[xiao-esp32-c6.json](https://github.com/user-attachments/files/15873857/xiao-esp32-c6.json)

GyverLibs, 2024-06-17:
пофиксил. Есть возможность протестить в железе?

[gio_esp32.zip](https://github.com/user-attachments/files/15874125/gio_esp32.zip)

GyverLibs, 2024-06-17:
видимо для другой архитектуры C версий у всех еспшек отличия в ядре

amirotin, 2024-06-17:
> так, и ещё раз [gio_esp32.zip](https://github.com/user-attachments/files/15874198/gio_esp32.zip)
Этот вариант так же работает

> видимо для другой архитектуры C версий у всех еспшек отличия в ядре
никакого стандарта у людей...))

GyverLibs, 2024-06-17:
ну всё пасиба, выкатываю патч тогда

ineu, 2025-02-18:
Здравствуйте. В 1.3.8 эта проблема вернулась, сломано в https://github.com/GyverLibs/GyverIO/commit/9b70e5ba27a55fe63ee9dee9b345e51751ad1a9f.

GyverLibs, 2025-02-18:
на каком МК?

GyverLibs, 2025-02-18:
Протестируйте в железе плиз

[gio_esp32.zip](https://github.com/user-attachments/files/18848190/gio_esp32.zip)

ineu, 2025-02-18:
> на каком МК?

esp32-c6, плата super mini

> Протестируйте в железе плиз

Закинул поверх 1.3.9, работает, спасибо!

GyverLibs, 2025-02-18:
О кайф, закину сегодня

closed 2025-02-18 by GyverLibs