GyverLibs/GyverIO #4: ESP32 error: jump to case label

creatormir, 2024-12-11:
Arduino IDE v 2.3.4
```
In file included from C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/esp32-hal.h:53:0,
                 from C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:35,
                 from C:\Users\user\AppData\Local\arduino\sketches\93875EE3202889434D2F73702F02DD6C\sketch\sketch_dec11a.ino.cpp:1:
C:\Users\user\Documents\Arduino\libraries\GyverIO-main\src/gio/gio_esp32.h: In function 'void gio::mode(uint8_t, uint8_t)':
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/esp32-hal-gpio.h:34:27: error: jump to case label [-fpermissive]
 #define OUTPUT            0x02
                           ^
C:\Users\user\Documents\Arduino\libraries\GyverIO-main\src/gio/gio_esp32.h:40:14: note: in expansion of macro 'OUTPUT'
         case OUTPUT:
              ^
In file included from C:\Users\user\Documents\Arduino\libraries\GyverIO-main\src/gio/gio.h:14:0,
                 from C:\Users\user\Documents\Arduino\libraries\GyverIO-main\src/GyverIO.h:3,
                 from C:\Users\user\Documents\Arduino\libraries\EncButton-main\src/core/io.h:3,
                 from C:\Users\user\Documents\Arduino\libraries\EncButton-main\src/core/VirtButton.h:5,
                 from C:\Users\user\Documents\Arduino\libraries\EncButton-main\src/core/Button.h:4,
                 from C:\Users\user\Documents\Arduino\libraries\EncButton-main\src/EncButton.h:4,
                 from C:\Users\user\AppData\Local\Temp\.arduinoIDE-unsaved20241111-2860-fiqxbx.k2zj8\sketch_dec11a\sketch_dec11a.ino:64:
C:\Users\user\Documents\Arduino\libraries\GyverIO-main\src/gio/gio_esp32.h:17:14: note:   crosses initialization of 'uint32_t rtc_reg'
     uint32_t rtc_reg(rtc_gpio_desc[pin].reg);                                                             \
              ^
C:\Users\user\Documents\Arduino\libraries\GyverIO-main\src/gio/gio_esp32.h:34:13: note: in expansion of macro '__esp32_rtc_idf3'
             __esp32_rtc_idf3(pin);
             ^
C:\Users\user\Documents\Arduino\libraries\GyverIO-main\src/gio/gio_esp32.h:52:9: error: jump to case label [-fpermissive]
         default:
         ^
C:\Users\user\Documents\Arduino\libraries\GyverIO-main\src/gio/gio_esp32.h:17:14: note:   crosses initialization of 'uint32_t rtc_reg'
     uint32_t rtc_reg(rtc_gpio_desc[pin].reg);                                                             \
              ^
C:\Users\user\Documents\Arduino\libraries\GyverIO-main\src/gio/gio_esp32.h:34:13: note: in expansion of macro '__esp32_rtc_idf3'
             __esp32_rtc_idf3(pin);
             ^

exit status 1

Compilation error: exit status 1
```

GyverLibs, 2024-12-11:
Понял принял. Рекомендую всё таки обновить ядро, версия видимо совсем древняя, 5-6 летней свежести)

creatormir, 2024-12-11:
Обновление ядра помогло

closed 2024-12-11 by creatormir