GyverLibs/GyverIO #6: Arduino IDE2 ESP32S3 gpio47 gpio48

DMG-115, 2025-01-26:
При компиляции в Arduino IDE2 не работет энкодер на gpio47 gpio48. 
Пробовал на версиях ядра ESP32 2.0.16, 2.0.17, 3.1.1. 
Причём на Platformio всё работает.
При замене фуннкций в файле "io.cpp" библилтеки "EncButton" на ардуиновские либо на еэспишные всё работает.
`#include "io.h"

bool __attribute__((weak)) EB_read(uint8_t pin) {
    //return gio::read(pin);
    //return digitalRead(pin);
    return gpio_get_level((gpio_num_t)pin);
}

void __attribute__((weak)) EB_mode(uint8_t pin, uint8_t mode) {
    //gio::init(pin, mode);
    //pinMode(pin, mode);
    gpio_set_direction((gpio_num_t)pin, GPIO_MODE_INPUT);
}

uint32_t __attribute__((weak)) EB_uptime() {
    return millis();
}`

GyverLibs, 2025-01-27:
А версия gyverio одинаковая и последняя в обеих ide?

DMG-115, 2025-01-27:
Да везде последняя версия библиотек "GyverIO" и "EncButton".

DMG-115, 2025-01-27:
Ok.

GyverLibs, 2025-01-27:
Насколько я вижу, в коде есть жесткое ограничение максимальным номером gpio 46. Это не мой код

GyverLibs, 2025-01-27:
а если вот так попробовать?

[gio_esp32.zip](https://github.com/user-attachments/files/18555384/gio_esp32.zip)

DMG-115, 2025-01-27:
Работает.

GyverLibs, 2025-01-27:
Огонь спасибо, теперь у нас самый правильный код) закидываю апдейт

closed 2025-01-27 by GyverLibs

DMG-115, 2025-01-27:
> Ну тогда это какой то баг. Не может на одинаковых версиях окружения работать по разному <
Platformio оказывается не обновляет "GyverIO" с гита, так что установлена была старая версия библиотеки.

DMG-115, 2025-01-27:
В апдейтах библиотек  Platformio "GyverIO" нет.

GyverLibs, 2025-01-27:
Платформио не обновляет, надо вручную делать запрос в консоль или в быстрых командах